tim-balloon/TIMflight #33: Make `mcp` fail gracefully if not hostnames are resolved

evanmayer, 2023-01-31:
**Describe the bug**
`mcp/communications/pilot.c:71` calls `initBITsender()`, defined in `./common/bitserver.c`. `bitserver.c:340` calls `gethostbyname()` on strings from the array pilot_target_names, defined in `blast_config/command_list.c:53`. This function relies on the `/etc/hosts` file mapping these hostnames to IP addresses. If the `/etc/hosts` file contains no entries for all hosts, `mcp` segfaults. It's not clear how many hosts must be in there for this to work.

**To Reproduce**
Steps to reproduce the behavior:
1. Comment out anything starting with 192.168.1.* from `/etc/hosts` on any computer of VM that can compile and run `mcp`.
2. Compile `mcp` for debug: `cd TIMflight/mcp/ && mkdir build && cd build && cmake .. -DCMAKE_BUILD_TYPE=Debug && cmake --build .`
3. Run `mcp`: `sudo ./mcp -x`
4. Observe segfault.

**Expected behavior**
`mcp` checks the return values for each host it tries, and fails gracefully with an error message if no hostnames are resolved. See other functions that do this with reporting that uses `berror` macro.

**Screenshots**
N/A

**Environment info (please complete the following information):**
 - PC hardware: vagrant or FC
 - OS: Debian 8 or Ubuntu 20.04
 - Software version: main commit SHA: `3d9a6562d32ca0f9fa86d15743d3871b96a549cc`

**Additional context**
N/A

evanmayer, 2023-01-31:
Related to #31

mxk62, 2023-02-14:
> It's not clear how many hosts must be in there for this to work.

Just as a side note, the hosts which currently need to be specified in `/etc/hosts` to prevent `mcp` from segfaulting are: `highbay`, `gollum`, `smeagol`, `galadriel` that is all hosts listed in `pilot_target_names` ([link](https://github.com/tim-balloon/TIMflight/blob/f5a21ff1fddffccb072b636d84e1102ebe08d9b1/blast_config/command_list.c#L53)).

closed 2023-04-11 by mxk62