tim-balloon/TIMflight #80: lat/lon from FC -> SCC

ianlowe13, 2024-02-26:
## Task

make sure there is a helper function that constantly updates the lat and lon in SC command packets and sends them out at a reasonable cadence - maybe 0.1 Hz?

## Acceptance

write the helper function and test from MCP that the SC computer is getting updated lat and lon data, this will be visible on the command line datastream as "received an update to LAT from FCx"

## Details

This function should go in the star_camera_transmit.c file and make use of the command data lat and lon fields that pointing.c updates for us

evanmayer, 2024-07-13:
@ianlowe13 is this done? If so, you can refer to a PR or commit in the blastcam repo and close this.

ianlowe13, 2024-07-14:
Finished in PR #85

closed 2024-07-14 by ianlowe13